koide3/interactive_slam #2: The loaded map disappears no result

mcc1710, 2019-10-29:
Hello  koide3,

thank you for the amazing work you have done with the hdl_graph_slam. Because there no motion compensation is integrated I wanted to correct the map with the interactive_slam.
The installation was successful but when I try to perform your first example (automatic loop detection) an error came up.

Steps I done:
- I load the hdl_400_dumped -> successfully (map and graph are visible)
- I open the automatic loop detection window and change the parameters as in the tutorial explained successfully (parameters can be changed)
- I start the automatic loop detection -> The error comes up
    -> The automatic loop detection starts and i see how the edges between the blue node and the      10 to 12 green nodes are generated 
    -> As soon as all of the green nodes (not the complete map -> 10 to 12 nodes) are connected to the blue node the map disappears and nothing happens
    -> In the terminal i see that some edges were calculated but after 10 to 12 calculations nothing happens anymore

My PC configuration:
RAM: 11,7 GiB
Processor: Intel® Core™ i7-2620M CPU @ 2.70GHz × 4 
Grafik: NVC3
Storage: 250,9 GB SSD

Can you please give me a hint how to solve this issue?

mcc1710, 2019-10-29:
[Interactive_SLAM_Video.zip](https://github.com/koide3/interactive_slam/files/3782303/Interactive_SLAM_Video.zip)

Here also an video of the error behavior

koide3, 2019-10-29:
Hi @mcc1710 ,
Thank you for reporting the issue. It seems I uploaded a wrong file. Could you try this: https://drive.google.com/open?id=1NqJwdkazMABVqryBUvV0CRXRuw9diMr8

In the previous example file, no pose vertices were fixed and the pose graph was shifted to somewhere during optimization. I fixed the very first frame in the example file so it stays around the origin. I also changed the optimizer to levenberg-marquardt since it is more robust to such an unfixed graph.

mcc1710, 2019-10-29:
Thanks for the fast reply.
The issue is now solved :)

closed 2019-10-29 by mcc1710