koide3/interactive_slam #6: Automatic loop detection make map rotate

liuyf5231, 2019-11-08:
hi koide3,
 I followed this example ,`https://github.com/koide3/interactive_slam/wiki/Example1`.
  but after automatic loop detection, the saved map rotate some degree.
[image]

koide3, 2019-11-08:
Hi @liuyf5231 ,
Please check if your map data has a fixed vertex:
```
Menubar -> Graph -> Graph editor -> Fixed vertices
```

If no vertices are fixed, you can right-click any keyframe and enable "Fixed" checkbox so that the map does not move during optimization.

liuyf5231, 2019-11-08:
> Hi @liuyf5231 ,
> Please check if your map data has a fixed vertex:
> 
> ```
> Menubar -> Graph -> Graph editor -> Fixed vertices
> ```
> 
> If no vertices are fixed, you can right-click any keyframe and enable "Fixed" checkbox so that the map does not move during optimization.

it works. thanks.

closed 2019-11-08 by liuyf5231