koide3/interactive_slam #8: how to merge two map generated by hdl graph slam?

liuyf5231, 2019-12-02:
Hi koide3, 
     I generated two map by hdl graph slam, I want to know how to merge them?
     These two map's init point is 0,0,0. if I merge them,the second map will go to the 0,0,0 point of the first map.

koide3, 2019-12-03:
Hi @liuyf5231 ,
This example explains how to merge multiple maps. If you have any further questions, let me know.
https://github.com/SMRT-AIST/interactive_slam/wiki/Example3#merge

UriyMihailov, 2021-11-30:
Hi @koide3,
I have the same problem. I created two maps using **hdl_graph_slam**, then I opened them in **interactive_slam**. Since they both started from zero point, I needed to move one of them, which I did through node estimation. I can visually see the map changing its position. However, the moment I use the `optimize()` function, the map that I moved goes back to zero point as if I hadn't made any changes at all. And if I try to create an edge between the maps after movements, both of the maps become completely broken. I think the problem lies within the `optimize()` function, but I just can't figure out where exactly. Maybe you, as a creator, would have any idea about this?