koide3/interactive_slam #10: portable-file-dialogs.h error

rrkmiao, 2019-12-06:
In member function ‘std::__cxx11::string pfd::internal::dialog::buttons_to_name(pfd::choice) const’:
/home/XXX/catkin_ws/src/interactive_slam/thirdparty/portable-file-dialogs/portable-file-dialogs.h:493:18: error: ‘choice’ is not a class, namespace, or enumeration
             case choice::ok_cancel: return "okcancel";
                  ^
/home/XXX/catkin_ws/src/interactive_slam/thirdparty/portable-file-dialogs/portable-file-dialogs.h:494:18: error: ‘choice’ is not a class, namespace, or enumeration
             case choice::yes_no: return "yesno";
In member function ‘std::__cxx11::string pfd::internal::dialog::get_icon_name(pfd::icon) const’:
/home/XXX/catkin_ws/src/interactive_slam/thirdparty/portable-file-dialogs/portable-file-dialogs.h:506:18: error: ‘icon’ is not a class, namespace, or enumeration
             case icon::warning: return "warning";
                  ^
/home/XXX/catkin_ws/src/interactive_slam/thirdparty/portable-file-dialogs/portable-file-dialogs.h:507:18: error: ‘icon’ is not a class, namespace, or enumeration
             case icon::error: return "error";

/home/XXX/catkin_ws/src/interactive_slam/src/hdl_graph_slam/plane_alignment_modal.cpp:93:67:   required from here
/usr/include/eigen3/Eigen/src/Core/DenseBase.h:41:34: error: invalid use of incomplete type ‘struct Eigen::internal::traits<float>’
 template<typename Derived> class DenseBase

/home/XXX/catkin_ws/src/interactive_slam/src/hdl_graph_slam/plane_alignment_modal.cpp:1:
/usr/include/eigen3/Eigen/src/Core/DenseBase.h:57:61: error: invalid use of incomplete type ‘struct Eigen::internal::traits<float>’
     typedef typename internal::traits<Derived>::StorageKind StorageKind;

/home/XXX/catkin_ws/src/interactive_slam/src/hdl_graph_slam/plane_alignment_modal.cpp:1:
/usr/include/eigen3/Eigen/src/Core/DenseBase.h:75:46: error: invalid use of incomplete type ‘struct Eigen::internal::traits<float>’
     typedef typename NumTraits<Scalar>::Real RealScalar;

and so on, all about "portable-file-dialogs.h" and eigen3
Working on ubuntu 16.04.
Thanks for sharing your work! Looking forward to your reply!

whydaydayup, 2019-12-06:
I am also woking on ubuntu 16.04 with ROS Kinetic, I also have the same problem!
Wish you can help fix the problem, Thanks!

koide3, 2019-12-08:
Hi guys,
I confirmed that the package can be built on Kinetic using LLVM toolchain (clang + lld). Try:
```bash
sudo apt install clang-6.0 lld-6.0
sudo update-alternatives --install /usr/bin/ld ld /usr/bin/ld.lld-6.0 50
cd ~/catkin_ws && CC=clang CXX=clang++ catkin_make -DCMAKE_BUILD_TYPE=Release
```

If you want to revert ld to the default one (ld.bfd) after building the package, do:
```bash
sudo update-alternatives --install /usr/bin/ld ld /usr/bin/ld.bfd 10
sudo update-alternatives --config ld
# select ld.bfd
```

I try to update the code so that it works with gcc and ld on kinetic. But, it's low priority and no ETA so far.

rrkmiao, 2019-12-17:
Thanks! I tried, but it still didn't work.

koide3, 2019-12-18:
@rrkmiao ,
Could you try to remove *deval* and *build* directories in *catkin_ws* before doing *catkin_make*?

WbHappy, 2020-01-16:
@koide3 I can confirm that this works, both **master** and **devel** branch, when deleting **build** and **devel** directories. 

So, using exactly those instructions as specified by @koide3, works perfectly :
```
sudo apt install clang-6.0 lld-6.0
sudo update-alternatives --install /usr/bin/ld ld /usr/bin/ld.lld-6.0 50
sudo update-alternatives --install /usr/bin/clang clang /usr/bin/clang-6.0 50
sudo update-alternatives --install /usr/bin/clang++ clang++ /usr/bin/clang++-6.0 50
cd ~/catkin_ws && rm -rf build devel    # be aware of that this command removes build and devel directories
CC=clang CXX=clang++ catkin_make -DCMAKE_BUILD_TYPE=Release
```

QiukuZ, 2020-02-23:
I dont understand why `cd ~/catkin_ws && rm -rf build devel` can fix this problem.
Although I did pass the build after adding this instruction

koide3, 2020-02-24:
@QiukuZ 
catkin determines the compiler to be used when you run catkin_make first time. To change the compiler you have to remove ```build``` directory to force catkin to clear and reconfigure the build parameters cache.