koide3/interactive_slam #15: error: GLSL 3.30 is not supported on Ubuntu 16.04 ROS Kinetic

QiukuZ, 2020-02-23:
Hey, I build successfully.But when i run 

`rosrun interactive_slam interactive_slam`

get:

error : failed to compile shader /home/qk/catkin_ws/src/interactive_slam/data/shader/rainbow.vert
0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

error : failed to compile shader /home/qk/catkin_ws/src/interactive_slam/data/shader/rainbow.frag
0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

error : failed to link program
error: linking with uncompiled/unspecialized shadererror: linking with uncompiled/unspecialized shader
construct solver: lm_var_cholmod
# Using CHOLMOD poseDim -1 landMarkDim -1 blockordering 0
done
error : failed to compile shader /home/qk/catkin_ws/src/interactive_slam/data/shader/rainbow.vert
0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

error : failed to compile shader /home/qk/catkin_ws/src/interactive_slam/data/shader/rainbow.frag
0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

error : failed to link program
error: linking with uncompiled/unspecialized shadererror: linking with uncompiled/unspecialized shader

What should i do？Thanks！

autoperson, 2020-02-23:
I also have this problem with Ubuntu 18.04 in VMware

QiukuZ, 2020-02-23:
> I also have this problem with Ubuntu 18.04 in VMware @koide3

I fixed this , I change some code in /src/imgui_application.cpp  :
`  // glfwWindowHint(GLFW_CONTEXT_VERSION_MAJOR, 3);
  // glfwWindowHint(GLFW_CONTEXT_VERSION_MINOR, 0);

  glfwWindowHint(GLFW_CONTEXT_VERSION_MAJOR, 3);
  glfwWindowHint(GLFW_CONTEXT_VERSION_MINOR, 3);
  glfwWindowHint(GLFW_OPENGL_PROFILE, GLFW_OPENGL_CORE_PROFILE);`

autoperson, 2020-02-23:
> I fixed this , I change some code in /src/imgui_application.cpp :
> ` // glfwWindowHint(GLFW_CONTEXT_VERSION_MAJOR, 3);`
> ` // glfwWindowHint(GLFW_CONTEXT_VERSION_MINOR, 0);`
>
> ` [...]

Thankyou!