koide3/interactive_slam #17: Eigen error during runtime

ShreyanshDarshan, 2020-03-11:
I was able to build the package successfully (apart from a few warnings) on Ubuntu 16.04 and ROS Kinetic. However when I run: 
```bash
rosrun interactive_slam interactive_slam
```
The following gets printed in the terminal and the execution stops:
```bash
construct solver: lm_var_cholmod
# Using CHOLMOD poseDim -1 landMarkDim -1 blockordering 0
done
interactive_slam: /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:128: Eigen::internal::plain_array<float, 16, 0, 32>::plain_array() [T = float, Size = 16, MatrixOrArrayOptions = 0, Alignment = 32]: Assertion `(reinterpret_cast<size_t>(array) & (31)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.
Aborted (core dumped)
```
Would really appreciate any help. Thanks!

koide3, 2020-03-16:
Hi @ShreyanshDarshan ,
Are you using self-built Eigen, g2o, or any other libraries?

ShreyanshDarshan, 2020-03-17:
Hi @koide3 ,
I have installed the following libraries from source:
- GL3W
- GLFW
- PCL
- g2o

Should I have not?

koide3, 2020-03-17:
We have received this kind of Eigen-related errors caused by self-built PCL and g2o. Can you try pre-built ones that can be installed via apt-get?

ShreyanshDarshan, 2020-05-13:
Using pre-built PCL solved the issue for me. Thanks a lot!

closed 2020-05-13 by ShreyanshDarshan

Jaqen81, 2021-11-09:
> Using pre-built PCL solved the issue for me. Thanks a lot!

Hi, did u use the following code to pre-build pcl? I deleted self-build pcl-1.8 compeletely and use the following code install libpcl-dev successfully but my pcl is not working 

```
sudo add-apt-repository ppa:v-launchpad-jochen-sprickerhof-de/pcl
sudo apt-get update
sudo apt-get install libpcl-dev pcl-tools
```

continue to run catkin_make, the error showed:

```
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake:48 (message):
  **PCL can not be found on this machine**
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake:772 (pcl_report_not_found)
  fast_gicp/CMakeLists.txt:18 (find_package)

```

thanks a lot