koide3/interactive_slam #20: how to correct map in this situation?

liuyf5231, 2020-04-28:
Hi koide3,
    I want to know how to correct my points map in this situation? The pillars drift a little in the below picture.
    
[image]
[image]

koide3, 2020-05-01:
Hi @liuyf5231 ,
In this case, you can add a few [manual loop closure constraints](https://github.com/SMRT-AIST/interactive_slam/wiki/Example3#manual-loop-closing-) to force the doubled pillars to be merged into one, and then [refine the pose constraints](https://github.com/SMRT-AIST/interactive_slam/wiki/Example3#pose-edge-refinement-) created by the automatic SLAM. You can also perform [automatic loop closing](https://github.com/SMRT-AIST/interactive_slam/wiki/Example3#automatic-loop-closing-) with a large distance threshold to improve the local consistency.

liuyf5231, 2020-05-01:
> Hi @liuyf5231 ,
> In this case, you can add a few [manual loop closure constraints](https://github.com/SMRT-AIST/interactive_slam/wiki/Example3#manual-loop-closing-) to force the doubled pillars to be [...]

Thanks, I will try it.

liuyf5231, 2020-05-03:
> Hi @liuyf5231 ,
> In this case, you can add a few [manual loop closure constraints](https://github.com/SMRT-AIST/interactive_slam/wiki/Example3#manual-loop-closing-) to force the doubled pillars to be [...]

Hi koide3,
when I choose two keyframes to make a loop constraint, graph optimization is not performed?!  The two pillars are not be merged into one.

[image]
[image]

closed 2020-05-06 by liuyf5231

abhi5691, 2020-06-01:
Hi @liuyf5231 , How did you manage to solve the issue? I am facing a similar challenge. When I add a loop constraint manually, the algorithm simply adds an additional edge between the nodes, leaving them approximately where they were, and hence the map is not really optimized.

[image]
See the long edges in top centre and lower center, both pair of nodes are supposed to be much closer together, and were so constrained. What am I missing here?

Any input will help me a lot, thanks.

ZeyingXuHuaWei, 2021-09-11:
In this case I think it is better to add a function such that we can do loop closure by picking two 3d points.  Anyone know how to achieve this?