koide3/interactive_slam #21: Ability to delete edge

tim-fan, 2020-05-20:
Hello, thanks for the great package!
I'm just wondering if there is a way to delete edges with this tool?
The map I import from hdl_graph_slam has some incorrect loop closures I'd like to delete. Additionally I sometimes manually add loop closures incorrectly, and I'd like a way to undo them.

Is there a way to delete these edges?

koide3, 2020-05-20:
The current implementation does not provide a way to delete an edge. But, we will definitely add this ability in the future version.

So far, you can try the devel branch that incorporates some new features, and one of them allows the user to replace an existing pose edge constraint. You can select two keyframes that are connected by a wrong edge and give a correct pose. Then, the existing constraint will be replaced with the new one.

tim-fan, 2020-05-20:
OK, thanks for the info.

In the meantime for the incorrect hdl_graph_slam edges, I've found I can manually delete them in the exported g2o file. But yes a 'click and delete edge' feature would be very handy :thumbsup::thumbsup:

koide3, 2020-05-28:
Preliminary implementation of deleting edge ability is now available on the devel branch.

ThanhNhann, 2020-10-13:
Thank for adding delete edge button, but when I check the function `void delete_edge(EdgeView::Ptr edge)` and comment the part find and erase `edges_view`, the code still run fine. Can you explain the role of this part in this function ? 
Thanks!

koide3, 2020-10-14:
Can you point the code with a permalink so that I can take a look at it?

ThanhNhann, 2020-10-14:
Thank for reply, the code is below!
https://github.com/SMRT-AIST/interactive_slam/blob/1ecd2f95d3bd8eee6a986c8e49a8745ab40003aa/include/hdl_graph_slam/view/interactive_graph_view.hpp#L93-L97

koide3, 2020-10-14:
```edges_view``` is a subset of ```drawables``` and used to manage right click menubar. Even if you comment out these lines, the deleted edge properly disappears as far as the following lines are preserved and the drawing object of the edge is removed from ```drawables``` (some unused ```edge_view``` objects would remain though). 

https://github.com/SMRT-AIST/interactive_slam/blob/1ecd2f95d3bd8eee6a986c8e49a8745ab40003aa/include/hdl_graph_slam/view/interactive_graph_view.hpp#L99-L103