koide3/interactive_slam #24: Adding a loop constraint doesn't lead to corresponding fix to the map

abhi5691, 2020-05-31:
Hello,
Thanks for creating this fabulous tool.

I am trying to fix a map I generated using hdl_graph_slam for a nearly 1.2 km x 0.7 km loop. The generated map had major aberrations along the z-axis, as also some deficiencies in xy plane.

I have used interactive_slam for some maps in past. Often, adding manual loop constraints shows significant fixes in the map. However at times, it simply adds an additional edge between the selected nodes while leaving them where they were (and hence the map stays as before: erratic) or sometimes just moving them partially.

I am sure there is a constraint conflicting the new loop addition, but a clearer idea might help me fix the map better.

abhi5691, 2020-05-31:
[image]

As visible, the long edge at top center is unintentional, the nodes are supposed to be close together, and were so constrained manually.

koide3, 2020-06-01:
Hi @abhi5691 ,
This behavior is caused by the robust kernel that prevents to largely change the estimate of the pose graph. Possible solutions are:
- choose "NONE" kernel when inserting a loop edge so that the edge will not be ignored
- perform graph optimization several times (```Main menu -> Graph -> Optimize```)

koide3, 2020-06-05:
I found that a recent update introduced a bug that disables pose graph optimization after manual loop closing. I fixed this bug and pushed an updated code that will be merged soon.

oscarmendezm, 2020-09-23:
This is also happening to me, even with the updated version. It seems to have something to do with floor constraints, as removing these makes the map immediately snap to the manual loop closure. Not sure what is going on.