koide3/interactive_slam #27: how to change the size of the displayed vertexes

Flandoe, 2020-07-04:
Shown as below, my vertexes is too dense to choose a single one. There is any way to change the displayed size of the vertexes. Thank you in advance.
[image]

koide3, 2020-07-06:
@Flandoe ,
I will make the sphere size changeable in the future. So far, you can try new SSAO rendering in the devel branch that enhances depth perception and allows to distinguish very close spheres.

[image]
[image]

Flandoe, 2020-07-06:
@koide3 I switched my branch to devel and recompiled. But there was no ssao in the shader setting window. Did I miss some steps?
[image]

koide3, 2020-07-06:
@Flandoe 
Apologize, it is available on ```ssao``` branch but not ```devel```.

Flandoe, 2020-07-06:
@koide3 I used ssao but it was still not ideal. Is there a way to change the sphere size by modifying the code.
[image]

koide3, 2020-07-06:
@Flandoe 
OK, you can change the sphere scale factor at the following line:

https://github.com/SMRT-AIST/interactive_slam/blob/1f07e3766afff6d73b985c410fe116a33134fc19/include/hdl_graph_slam/view/keyframe_view.hpp#L48

closed 2020-07-06 by Flandoe