koide3/interactive_slam #28: Click New map and no responced

whu-lyh, 2020-07-13:
I build succeed in ubuntu18.04 and a docker environment. I run 
`rosrun interactive_slam interactive_slam`
the ui shown, but when i click the  New map, there is no dialog to choose the bag or something else.
Meanwhile i find the console occurs an error like this:
```
glfw error 65544: X11: RandR gamma ramp support seems broken
glfw error 65544: Linux: Failed to watch for joystick connections in /dev/input: No such file or directory
glfw error 65544: Linux: Failed to open joystick device directory /dev/input: No such file or directory
construct solver: lm_var_cholmod
# Using CHOLMOD poseDim -1 landMarkDim -1 blockordering 1
done
```
So, anybody who meet this problem? Thanks guys.

koide3, 2020-07-15:
Did you run the program on docker? I'm not sure, but the file dialog library may have trouble with docker.

https://github.com/samhocevar/portable-file-dialogs

whu-lyh, 2020-07-16:
Well，i did run interactive_slam in docker, may be i should build without docker condition

closed 2020-07-16 by whu-lyh

asbe, 2020-10-02:
Did you successfully resolve this? I am facing the same issue, more or less, having built on Ubuntu 18.04 under WSL2.

awesomebytes, 2021-07-28:
For anyone that reaches this problem using it in linux inside of Docker, a valid workaround is to install in Docker:
```bash
sudo apt-get install zenity
```