koide3/interactive_slam #29: How to correct map using vertical walls?

zxl19, 2020-07-14:
Hi, thank you for your work!

I tried to add constraints to the pose graph by vertical walls. I followed the instructions in [Example3](https://github.com/SMRT-AIST/interactive_slam/wiki/Example3), but it seems that some pictures are missing in the tutorial in the *Plane Constraints* section. I tried `Right click a point -> Plane detection`, but the plane I segmented(the orange dots) includes a large part of the ground:

[image]

Could you please elaborate the process about adding **vertical planes** as vertices in the pose graph? Thanks!

Sincerely.

koide3, 2020-07-15:
You can detect vertical planes in the same way, but region growing parameters (in particular ```Smoothness threshold``` and ```Curvature threshold```) have to be changed such that orange points appear on only the plane you want to detect. 

The current plane detection implementation is naive and sometimes suffer from this problem. I implemented a more better way to detect planes in the closed source version, and it would be merged into this open source version in the future.

zxl19, 2020-07-15:
Thank you for your reply!
I'll change the parameters to see if the result gets any better.
Also, I'm wondering if there are any ways to export poses of the ego vehicle so that I can compare pose estimates before and after manual correction with groundtruth to see the changes of mapping precision.

Best regards.

koide3, 2020-07-16:
You can find ```graph.g2o``` in a saved map directory that describes the keyframe poses in the pose graph.

closed 2020-07-16 by zxl19