koide3/interactive_slam #30: About vertical drift and revoking operations

zxl19, 2020-07-18:
Hi, thank you for your excellent work!
I'm using LeGO-LOAM to generate odometry data of an industrial plant and correcting the map using interactive_slam afterwards. Due to the scale of the map, drift in the z axis is aggregated and becoming severe(~10m) at the end of the vehicle path. I tried manually adding loop closures, but it only corrects mapping error locally. Later I tried adding plane constraints, but it does little help because without knowing the groundtruth I can only assume that the plane is parallel to the start ground. Also sometimes if I set the prior of two planes to be identical, the map will become warped, causing severe structual error. I can delete the wrong edges but the map doesn't resume to its shape before.
1. Are there any empirical methods to correct map drift in the z axis? I tried planar ground hypothesis but it seems to be too strong for most scenarios. I'm considering spliting the rosbag into several segments and register the segmented map in interactive_slam to build an entire map. Is that going to help? 
2. Are there any ways to revoke an operation and undo the changes to the map in interactive_slam? Like `Ctrl + Z` in other scenarios. 

Best regards.

koide3, 2020-07-21:
1. It is still a difficult situation, and splitting the rosbag would not help so much. If there are walls, it would be good to add them to the map. By injecting information that the walls are perpendicular w.r.t. the floor (and parallel w.r.t. each other), the z-axis drift may be implicitly corrected. It is also worth considering to use the gravity vector if you can use an IMU. 

2. After removing an edge, you need to perform pose graph optimization manually (```Menubar -> Graph -> Optimize```), then the map should be reverted to the previous state. I want to implement some command history and undo functions someday, but I don't have enough spare time so far...

zxl19, 2020-07-21:
I'll try that, thanks!

closed 2020-07-21 by zxl19