koide3/interactive_slam #35: build error on Ubuntu 18.04 & ROS melodic

XuanboWang, 2020-08-19:
`Linking CXX executable /home/wangboxuan/Desktop/map_ws/devel/lib/interactive_slam/interactive_slam
CMakeFiles/interactive_slam.dir/src/hdl_graph_slam/plane_detection_window.cpp.o:(.data.rel.ro._ZTVN3pcl25SampleConsensusModelPlaneINS_9PointXYZIEEE[_ZTVN3pcl25SampleConsensusModelPlaneINS_9PointXYZIEEE]+0x28): undefined reference to `pcl::SampleConsensusModelPlane<pcl::PointXYZI>::computeModelCoefficients(std::vector<int, std::allocator<int> > const&, Eigen::Matrix<float, -1, 1, 0, -1, 1>&) const'
CMakeFiles/interactive_slam.dir/src/hdl_graph_slam/plane_detection_window.cpp.o:(.data.rel.ro._ZTVN3pcl25SampleConsensusModelPlaneINS_9PointXYZIEEE[_ZTVN3pcl25SampleConsensusModelPlaneINS_9PointXYZIEEE]+0x30): undefined reference to `pcl::SampleConsensusModelPlane<pcl::PointXYZI>::optimizeModelCoefficients(std::vector<int, std::allocator<int> > const&, Eigen::Matrix<float, -1, 1, 0, -1, 1> const&, Eigen::Matrix<float, -1, 1, 0, -1, 1>&) const'
CMakeFiles/interactive_slam.dir/src/hdl_graph_slam/plane_detection_window.cpp.o:(.data.rel.ro._ZTVN3pcl25SampleConsensusModelPlaneINS_9PointXYZIEEE[_ZTVN3pcl25SampleConsensusModelPlaneINS_9PointXYZIEEE]+0x38): undefined reference to `pcl::SampleConsensusModelPlane<pcl::PointXYZI>::getDistancesToModel(Eigen::Matrix<float, -1, 1, 0, -1, 1> const&, std::vector<double, std::allocator<double> >&) const'
CMakeFiles/interactive_slam.dir/src/hdl_graph_slam/plane_detection_window.cpp.o:(.data.rel.ro._ZTVN3pcl25SampleConsensusModelPlaneINS_9PointXYZIEEE[_ZTVN3pcl25SampleConsensusModelPlaneINS_9PointXYZIEEE]+0x48): undefined reference to `pcl::SampleConsensusModelPlane<pcl::PointXYZI>::countWithinDistance(Eigen::Matrix<float, -1, 1, 0, -1, 1> const&, double) const'
CMakeFiles/interactive_slam.dir/src/hdl_graph_slam/plane_detection_window.cpp.o:(.data.rel.ro._ZTVN3pcl25SampleConsensusModelPlaneINS_9PointXYZIEEE[_ZTVN3pcl25SampleConsensusModelPlaneINS_9PointXYZIEEE]+0x50): undefined reference to `pcl::SampleConsensusModelPlane<pcl::PointXYZI>::projectPoints(std::vector<int, std::allocator<int> > const&, Eigen::Matrix<float, -1, 1, 0, -1, 1> const&, pcl::PointCloud<pcl::PointXYZI>&, bool) const'
CMakeFiles/interactive_slam.dir/src/hdl_graph_slam/plane_detection_window.cpp.o:(.data.rel.ro._ZTVN3pcl25SampleConsensusModelPlaneINS_9PointXYZIEEE[_ZTVN3pcl25SampleConsensusModelPlaneINS_9PointXYZIEEE]+0x58): undefined reference to `pcl::SampleConsensusModelPlane<pcl::PointXYZI>::doSamplesVerifyModel(std::set<int, std::less<int>, std::allocator<int> > const&, Eigen::Matrix<float, -1, 1, 0, -1, 1> const&, double) const'
collect2: error: ld returned 1 exit status
interactive_slam/CMakeFiles/interactive_slam.dir/build.make:970: recipe for target '/home/wangboxuan/Desktop/map_ws/devel/lib/interactive_slam/interactive_slam' failed
make[2]: *** [/home/wangboxuan/Desktop/map_ws/devel/lib/interactive_slam/interactive_slam] Error 1
CMakeFiles/Makefile2:4845: recipe for target 'interactive_slam/CMakeFiles/interactive_slam.dir/all' failed
make[1]: *** [interactive_slam/CMakeFiles/interactive_slam.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2`

As we can see, it reminds us that cmake can't find the PCL SampleConsensusModelPlane function in the link phase. I've installed PCL 1.8 and so many dependence libs, but i can't solve this problem. I'm not familiar with PCL, I want to know what's wrong.Thanks.

Jaqen81, 2021-11-16:
To check if your pcl is correctly installed, you could use pcl-viewer to check a .pcl file

doche, 2022-11-04:
hi, how to install imgui lib, thanks.