koide3/interactive_slam #40: detect plane fail

chenxianbo, 2020-12-09:
Hi，thank for your work。
       I test to detect plane many times，but scucces only sevral times，it's always printf “two points for region growing”。（Some plane region never scucess ！！！）
      I add some prints in the code， find not a keyfeame find one or more neighboor points around the select point， the accumulated points num is zero， I was confused。 I enlarge the radius 20~30m,  it can find neighboor points； 1~5m,  failed。
     All the operation is in centrair_graph dataset， detect plane region is as same as Example3 show。

koide3, 2020-12-10:
At this moment, the plane detection algorithm is very naive, and we need to fine-tune some parameters depending on the environment. To detect large walls in ```centrainr_graph``` dataset, try to increase ```Initial neighbor radius``` and ```Normal estimation radius``` to 5.0 m.

chenxianbo, 2020-12-10:
> At this moment, the plane detection algorithm is very naive, and we need to fine-tune some parameters depending on the environment. To detect large walls in `centrainr_graph` dataset, try to increase `Initial neighbor radius` and `Normal estimation radius` to 5.0 m.

chenxianbo, 2020-12-10:
> At this moment, the plane detection algorithm is very naive, and we need to fine-tune some parameters depending on the environment. To detect large walls in `centrainr_graph` dataset, try to increase `Initial neighbor radius` and `Normal estimation radius` to 5.0 m.

I  find I click the same place‘s point,  the point's coordinate is  different very much when  zoom  in  and zoom out。

koide3, 2020-12-14:
> I find I click the same place‘s point, the point's coordinate is different very much when zoom in and zoom out。And when rotate the screen ， the same point show very different coordinate

I couldn't reproduce this phenomenon. Can you re-check if you clicked the same point?

> Another question， the keyframe pose graph is always roatate the opposite direction with the point cloud。

I don't understand what is happening on your environment. Is it a kind of visualization issues?

chenxianbo, 2020-12-14:
@koide3 
My environment is ubuntu16.04 and kenitic，I am sure I select the same cloud point， when I roate the screen， it will show big different coordinate。But the pose graph have no problem。

chenxianbo, 2020-12-14:
@I try ubuntu18.04 and melodic， it is the same， I down the newest master branch。
[image]
[image]
[image]

koide3, 2020-12-15:
It looks something wrong on rendering. Can you check the OpenGL version on your PC with ```glxinfo | grep OpenGL```?

chenxianbo, 2020-12-15:
@koide3 
xianbo@ubuntu:~/InteractiveSlam$ glxinfo | grep OpenGL
OpenGL vendor string: VMware, Inc.
OpenGL renderer string: SVGA3D; build: RELEASE;  LLVM;
OpenGL core profile version string: 3.3 (Core Profile) Mesa 20.0.8
OpenGL core profile shading language version string: 3.30
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 3.3 (Compatibility Profile) Mesa 20.0.8
OpenGL shading language version string: 3.30
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 2.0 Mesa 20.0.8
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 1.0.16
OpenGL ES profile extensions:

koide3, 2020-12-17:
I've not tried to run this program on a virtual machine, and I guess it has some issues relating to the virtual GL driver. Can you try to run it on a non-virtual environment?