koide3/interactive_slam #57: Export PointCloud have problem

1003613248, 2021-11-05:
when i click the Export PointCloud button ，then jump out the file directory， i set the store directory . when i click OK， there is no response at all. i have tried several times , still have problem.

1003613248, 2021-11-05:
i run the Interactive SLAM on ubuntu18.04, physical machine

Jaqen81, 2021-11-08:
hi，have u had"Segmentation fault (core dumped)" error? I've installed all the dependencies and compiled it successfully, still couldn't run the GUI, thanks

1003613248, 2021-11-08:
dear Jaqen81 ,i have no this problem that you said. In the baginning , i also can not run the GUI, but i run  export MESA_GL_VERSION_OVERRIDE=3.3, it works.

1003613248, 2021-11-08:
I have resolved the export pointcloud problem. when you select the store path, you must select a pcd file, then click OK button.

closed 2021-11-08 by 1003613248

Jaqen81, 2021-11-08:
> dear Jaqen81 ,i have no this problem that you said. In the baginning , i also can not run the GUI, but i run export MESA_GL_VERSION_OVERRIDE=3.3, it works.

Thanks for your reply, I just tried your method but still Segmentation fault, I'll find other way, thank you tho!