koide3/interactive_slam #62: build error

yijing2050, 2022-03-21:
/home/yijing/pakagegit/interactive_slam/devel/lib/libimgui.a(imgui_demo.cpp.o):imgui_demo.cpp:(.text+0xe291): 跟着更多未定义的参考到 ImGui::TableNextColumn()
/home/yijing/pakagegit/interactive_slam/devel/lib/libimgui.a(imgui_demo.cpp.o)：在函数‘ShowDemoWindowTables()’中：
imgui_demo.cpp:(.text+0xe412)：对‘ImGui::TableSetColumnIndex(int)’未定义的引用
imgui_demo.cpp:(.text+0xe47f)：对‘ImGui::TableSetBgColor(int, unsigned int, int)’未定义的引用
/home/yijing/pakagegit/interactive_slam/devel/lib/libimgui.a(imgui_demo.cpp.o)：在函数‘ShowDemoWindowWidgets()’中：
imgui_demo.cpp:(.text+0x1335f)：对‘ImGui::BeginTable(char const*, int, int, ImVec2 const&, float)’未定义的引用
imgui_demo.cpp:(.text+0x1339b)：对‘ImGui::TableNextColumn()’未定义的引用
imgui_demo.cpp:(.text+0x133c7)：对‘ImGui::EndTable()’未定义的引用
imgui_demo.cpp:(.text+0x133f5)：对‘ImGui::BeginTable(char const*, int, int, ImVec2 const&, float)’未定义的引用
imgui_demo.cpp:(.text+0x1343d)：对‘ImGui::TableNextRow(int, float)’未定义的引用
imgui_demo.cpp:(.text+0x13442)：对‘ImGui::TableNextColumn()’未定义的引用
imgui_demo.cpp:(.text+0x1346b)：对‘ImGui::TableNextColumn()’未定义的引用
imgui_demo.cpp:(.text+0x1347e)：对‘ImGui::TableNextColumn()’未定义的引用
imgui_demo.cpp:(.text+0x13497)：对‘ImGui::EndTable()’未定义的引用
/home/yijing/pakagegit/interactive_slam/devel/lib/libimgui.a(imgui_demo.cpp.o)：在函数‘ImGui::ShowDemoWindow(bool*)’中：
imgui_demo.cpp:(.text+0x1924f)：对‘ImGui::BeginTable(char const*, int, int, ImVec2 const&, float)’未定义的引用
imgui_demo.cpp:(.text+0x19273)：对‘ImGui::EndTable()’未定义的引用
imgui_demo.cpp:(.text+0x19a5f)：对‘ImGui::BeginTable(char const*, int, int, ImVec2 const&, float)’未定义的引用
imgui_demo.cpp:(.text+0x19a6c)：对‘ImGui::TableNextColumn()’未定义的引用
imgui_demo.cpp:(.text+0x19a84)：对‘ImGui::TableNextColumn()’未定义的引用
imgui_demo.cpp:(.text+0x19a9c)：对‘ImGui::TableNextColumn()’未定义的引用
imgui_demo.cpp:(.text+0x19ab4)：对‘ImGui::TableNextColumn()’未定义的引用
imgui_demo.cpp:(.text+0x19acc)：对‘ImGui::TableNextColumn()’未定义的引用
/home/yijing/pakagegit/interactive_slam/devel/lib/libimgui.a(imgui_demo.cpp.o):imgui_demo.cpp:(.text+0x19ae4): 跟着更多未定义的参考到 ImGui::TableNextColumn()
/home/yijing/pakagegit/interactive_slam/devel/lib/libimgui.a(imgui_demo.cpp.o)：在函数‘ImGui::ShowDemoWindow(bool*)’中：
imgui_demo.cpp:(.text+0x19b5c)：对‘ImGui::EndTable()’未定义的引用
/home/yijing/pakagegit/interactive_slam/devel/lib/libimgui.a(imgui_widgets.cpp.o)：在函数‘ImGui::SeparatorEx(int)’中：
imgui_widgets.cpp:(.text+0x9c54)：对‘ImGui::PushColumnsBackground()’未定义的引用
imgui_widgets.cpp:(.text+0x9ccb)：对‘ImGui::PopColumnsBackground()’未定义的引用
/home/yijing/pakagegit/interactive_slam/devel/lib/libimgui.a(imgui_widgets.cpp.o)：在函数‘ImGui::Separator()’中：
imgui_widgets.cpp:(.text+0x9f74)：对‘ImGui::PushColumnsBackground()’未定义的引用
imgui_widgets.cpp:(.text+0x9ff9)：对‘ImGui::PopColumnsBackground()’未定义的引用
/home/yijing/pakagegit/interactive_slam/devel/lib/libimgui.a(imgui_widgets.cpp.o)：在函数‘ImGui::Selectable(char const*, bool, int, ImVec2 const&)’中：
imgui_widgets.cpp:(.text+0x15fd8)：对‘ImGui::PushColumnsBackground()’未定义的引用
imgui_widgets.cpp:(.text+0x1628e)：对‘ImGui::PopColumnsBackground()’未定义的引用
imgui_widgets.cpp:(.text+0x164ef)：对‘ImGui::TablePopBackgroundChannel()’未定义的引用
imgui_widgets.cpp:(.text+0x1650f)：对‘ImGui::TablePushBackgroundChannel()’未定义的引用
collect2: error: ld returned 1 exit status
interactive_slam/CMakeFiles/interactive_slam.dir/build.make:977: recipe for target '/home/yijing/pakagegit/interactive_slam/devel/lib/interactive_slam/interactive_slam' failed
make[2]: *** [/home/yijing/pakagegit/interactive_slam/devel/lib/interactive_slam/interactive_slam] Error 1
CMakeFiles/Makefile2:5061: recipe for target 'interactive_slam/CMakeFiles/interactive_slam.dir/all' failed
make[1]: *** [interactive_slam/CMakeFiles/interactive_slam.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j16 -l16" failed
I don't know why. Is it right when we download thirdparty of Dear IMGUI and we can just use it without installing it?

doche, 2022-11-04:
me too,how to install imgui lib, thanks..

doche, 2022-11-07:
hi  [yijing2050]:
      have found the right way ? thanks..

yijing2050, 2022-11-07:
Yes,I have.You should download the right edition of it.

doche, 2022-11-07:
> Yes,I have.You should download the right edition of it.
> […](#)

the right imgui？ the master version of  imgui is wrong? which one is right?

yijing2050, 2022-11-07:
[image]

yijing2050, 2022-11-07:
c077dd4 is the right edition of imgui!

doche, 2022-11-07:
> c077dd4 is the right edition of imgui!
> 
> > > Yes,I have.You should download the right edition of it.
> > > […](#)
> > 
> > 
> > the right imgui？ the master version of imgui is wrong? which one is right?

have compiled successfully,thanks a lot .   how to get the [hdl_400_dumped.tar.gz] ?