koide3/interactive_slam #63: Can not open file browser dialog when start interactive slam in docker noetic

sonphambk, 2022-06-16:
Hi authors
I have a problem when start interactive slam in docker-container.
After running command `rosrun interactive_slam interactive_slam`. The window show up but when I open map from menubar. The open file dialog can not open. 
Do you have a check about this problem?
Thanks

dbaldzhiev, 2023-05-04:
I have the same issue. Are you running the docker in Windows?

christophTUM, 2023-05-18:
Did you fix it @dbaldzhiev @sonphambk ?

sonphambk, 2023-05-19:
Hi @christophTUM 
my solution is that copy build forder in docker container to local machine and you can start it out docker.

christophTUM, 2023-05-19:
I fixed it with https://github.com/SMRT-AIST/interactive_slam/issues/28

So you can close this issue.

closed 2023-05-19 by sonphambk