koide3/interactive_slam #74: How to view the large map without out of core memory

sonphambk, 2023-03-30:
Hi @kkoide3 @koide3, thank you for your work
I want to ask you in interactive-slam tool. How can we visualize the large point cloud with other data structure(octree) and edit the pose graph?
I am facing the challenge of large point cloud visualization. How do you think about this?
Thanks

sonphambk, 2023-04-13:
I means how we can visualize large point cloud without load whole into memory like potree and view based on LOD structure.
How do you think ? @koide3

cinogul, 2024-04-27:
Hi @koide3,

Thanks for this useful package. I've got a similar problem with poor performance on relatively large maps. I'm running WSL2 Ubuntu 20.04 with noetic. As for hardware I've got an AMD Ryzen 7 3700X 8-Core Processor, 3600 Mhz, 8 Core(s), with 16 Logical Processor(s), 16GB of ram, and a 3060 TI graphics card but I'm still only running the application at around 4 fps on average, which makes it quite difficult to edit the map. I've ensured that I have the right drivers installed and that the virtual machine has enough processors (12 cores) and memory (12gb + 2 swap). Is this performance issue just because of the usage of WSL or is there something I missed? Do you reckon that performance will improve if I just get a dual boot on my PC instead of using WSL?