koide3/interactive_slam #80: Ubuntu 20.04 --> build error "catkin_make -DCMAKE_BUILD_TYPE=Release"

mathewsarbuckle, 2024-08-03:
The actual error is extremely long (can't fit all of it here) but it mostly tied with pcl. Any assistance would be greatly appreicated!

beginning of error: 
[ 15%] Built target nav_msgs_generate_messages_eus
[ 16%] Building CXX object odometry_saver/CMakeFiles/odometry_saver.dir/src/odometry_saver.cpp.o
In file included from /usr/include/pcl-1.10/pcl/pcl_macros.h:77,
                 from /usr/include/pcl-1.10/pcl/PCLHeader.h:10,
                 from /usr/include/pcl-1.10/pcl/point_cloud.h:47,
                 from /opt/ros/noetic/include/pcl_ros/point_cloud.h:5,
                 from /home/mat/catkin_ws/src/odometry_saver/src/odometry_saver.cpp:12:
/usr/include/pcl-1.10/pcl/pcl_config.h:7:4: error: #error PCL requires C++14 or above
    7 |   #error PCL requires C++14 or above
      |    ^~~~~
In file included from /usr/include/pcl-1.10/pcl/console/print.h:44,
                 from /usr/include/pcl-1.10/pcl/conversions.h:53,
                 from /opt/ros/noetic/include/pcl_ros/point_cloud.h:8,
                 from /home/mat/catkin_ws/src/odometry_saver/src/odometry_saver.cpp:12:
/usr/include/pcl-1.10/pcl/pcl_config.h:7:4: error: #error PCL requires C++14 or above
    7 |   #error PCL requires C++14 or above
      |    ^~~~~
In file included from /opt/ros/noetic/include/pcl_ros/point_cloud.h:284,
                 from /home/mat/catkin_ws/src/odometry_saver/src/odometry_saver.cpp:12:
/usr/include/pcl-1.10/pcl/pcl_config.h:7:4: error: #error PCL requires C++14 or above
    7 |   #error PCL requires C++14 or above
      |    ^~~~~
^Cmake[2]: *** [odometry_saver/CMakeFiles/odometry_saver.dir/build.make:63: odometry_saver/CMakeFiles/odometry_saver.dir/src/odometry_saver.cpp.o] Interrupt
make[1]: *** [CMakeFiles/Makefile2:3004: odometry_saver/CMakeFiles/odometry_saver.dir/all] Interrupt
make: *** [Makefile:141: all] Interrupt
Traceback (most recent call last):
  File "/opt/ros/noetic/bin/catkin_make", line 306, in <module>
    sys.exit(main())
  File "/opt/ros/noetic/bin/catkin_make", line 249, in main
    run_command(cmd, make_path)
  File "/opt/ros/noetic/lib/python3/dist-packages/catkin/builder.py", line 241, in run_command
    proc.wait()
  File "/usr/lib/python3.8/subprocess.py", line 1083, in wait
    return self._wait(timeout=timeout)
  File "/usr/lib/python3.8/subprocess.py", line 1806, in _wait
    (pid, sts) = self._try_wait(0)
  File "/usr/lib/python3.8/subprocess.py", line 1764, in _try_wait
    (pid, sts) = os.waitpid(self.pid, wait_flags)

........................................................................................................

end of error: 
/usr/include/pcl-1.10/pcl/common/impl/copy_point.hpp:80:54: error: ‘enable_if_t’ is not a member of ‘std’; did you mean ‘enable_if’?
   80 |     struct CopyPointHelper<PointInT, PointOutT, std::enable_if_t<std::is_same<PointInT, PointOutT>::value>>
      |                                                      ^~~~~~~~~~~
      |                                                      enable_if
/usr/include/pcl-1.10/pcl/common/impl/copy_point.hpp:80:101: error: type/value mismatch at argument 3 in template parameter list for ‘template<class PointInT, class PointOutT, class Enable> struct pcl::detail::CopyPointHelper’
   80 | ntHelper<PointInT, PointOutT, std::enable_if_t<std::is_same<PointInT, PointOutT>::value>>
      |                                                                                   ^~~~~

/usr/include/pcl-1.10/pcl/common/impl/copy_point.hpp:80:101: note:   expected a type, got ‘(<expression error> < std::is_same< <template-parameter-1-1>, <template-parameter-1-2> >::value)’
/usr/include/pcl-1.10/pcl/common/impl/copy_point.hpp:80:106: error: expected unqualified-id before ‘>’ token
   80 | per<PointInT, PointOutT, std::enable_if_t<std::is_same<PointInT, PointOutT>::value>>
      |                                                                                   ^~

/usr/include/pcl-1.10/pcl/common/impl/copy_point.hpp:90:33: error: ‘enable_if_t’ is not a member of ‘std’; did you mean ‘enable_if’?
   90 |                            std::enable_if_t<boost::mpl::and_<boost::mpl::not_<std::is_same<PointInT, PointOutT>>,
      |                                 ^~~~~~~~~~~
      |                                 enable_if
/usr/include/pcl-1.10/pcl/common/impl/copy_point.hpp:90:33: error: ‘enable_if_t’ is not a member of ‘std’; did you mean ‘enable_if’?
   90 |                            std::enable_if_t<boost::mpl::and_<boost::mpl::not_<std::is_same<PointInT, PointOutT>>,
      |                                 ^~~~~~~~~~~
      |                                 enable_if
In file included from /usr/include/pcl-1.10/pcl/common/copy_point.h:58,
                 from /usr/include/pcl-1.10/pcl/common/impl/io.hpp:45,
                 from /usr/include/pcl-1.10/pcl/common/io.h:586,
                 from /usr/include/pcl-1.10/pcl/io/file_io.h:41,
[... 156 lines cut ...]
                 from /usr/include/c++/9/array:39,
                 from /usr/include/c++/9/tuple:39,
                 from /usr/include/c++/9/mutex:38,
                 from /home/mat/catkin_ws/src/odometry_saver/src/odometry_saver.cpp:1:
/usr/include/c++/9/bits/predefined_ops.h: In instantiation of ‘bool __gnu_cxx::__ops::_Iter_pred<_Predicate>::operator()(_Iterator) [with _Iterator = __gnu_cxx::__normal_iterator<const pcl::PCLPointField*, std::vector<pcl::PCLPointField> >; _Predicate = pcl::getFieldIndex(const pcl::PCLPointCloud2&, const string&)::<lambda(int)>]’:
/usr/include/c++/9/bits/stl_algo.h:120:14:   required from ‘_RandomAccessIterator std::__find_if(_RandomAccessIterator, _RandomAccessIterator, _Predicate, std::random_access_iterator_tag) [with _RandomAccessIterator = __gnu_cxx::__normal_iterator<const pcl::PCLPointField*, std::vector<pcl::PCLPointField> >; _Predicate = __gnu_cxx::__ops::_Iter_pred<pcl::getFieldIndex(const pcl::PCLPointCloud2&, const string&)::<lambda(int)> >]’
/usr/include/c++/9/bits/stl_algo.h:161:23:   required from ‘_Iterator std::__find_if(_Iterator, _Iterator, _Predicate) [with _Iterator = __gnu_cxx::__normal_iterator<const pcl::PCLPointField*, std::vector<pcl::PCLPointField> >; _Predicate = __gnu_cxx::__ops::_Iter_pred<pcl::getFieldIndex(const pcl::PCLPointCloud2&, const string&)::<lambda(int)> >]’
/usr/include/c++/9/bits/stl_algo.h:3969:28:   required from ‘_IIter std::find_if(_IIter, _IIter, _Predicate) [with _IIter = __gnu_cxx::__normal_iterator<const pcl::PCLPointField*, std::vector<pcl::PCLPointField> >; _Predicate = pcl::getFieldIndex(const pcl::PCLPointCloud2&, const string&)::<lambda(int)>]’
/usr/include/pcl-1.10/pcl/common/io.h:65:77:   required from here
/usr/include/c++/9/bits/predefined_ops.h:283:11: error: no match for call to ‘(pcl::getFieldIndex(const pcl::PCLPointCloud2&, const string&)::<lambda(int)>) (const pcl::PCLPointField&)’
  283 |  { return bool(_M_pred(*__it)); }
      |           ^~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/pcl-1.10/pcl/io/file_io.h:41,
                 from /usr/include/pcl-1.10/pcl/io/pcd_io.h:44,
                 from /opt/ros/noetic/include/pcl_conversions/pcl_conversions.h:70,
                 from /opt/ros/noetic/include/pcl_ros/point_cloud.h:9,
                 from /home/mat/catkin_ws/src/odometry_saver/src/odometry_saver.cpp:12:
/usr/include/pcl-1.10/pcl/common/io.h:65:9: note: candidate: ‘pcl::getFieldIndex(const pcl::PCLPointCloud2&, const string&)::<lambda(int)>’
   65 |         [&field_name](const auto field) { return field.name == field_name; });
      |         ^
/usr/include/pcl-1.10/pcl/common/io.h:65:9: note:   no known conversion for argument 1 from ‘const pcl::PCLPointField’ to ‘int’
make[2]: *** [odometry_saver/CMakeFiles/odometry_saver.dir/build.make:63: odometry_saver/CMakeFiles/odometry_saver.dir/src/odometry_saver.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:3004: odometry_saver/CMakeFiles/odometry_saver.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j1 -l1" failed

koide3, 2024-08-05:
It seems that there are some issues about the C++ version. Can you try adding `set(CMAKE_CXX_STANDARD 17)` in `CMakeLists.txt`?

mathewsarbuckle, 2024-08-05:
Hi, I tried set(CMAKE_CXX_STANDARD 17) and set(CMAKE_CXX_STANDARD 14) in my CMakeLists.txt. I'm still running into the same error with c++ version.

mathewsarbuckle, 2024-08-05:
I got past the build error, thank you! I'm running into a new error when I try to launch the application. It has something to do with shader version? Any insight would be greatly appreciated, thanks!

Error: 
mat@mat-VirtualBox:~/catkin_ws$ rosrun interactive_slam interactive_slam
error : failed to compile shader /home/mat/catkin_ws/src/interactive_slam/data/shader/rainbow.vert
0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, 1.00 ES, and 3.00 ES

error : failed to compile shader /home/mat/catkin_ws/src/interactive_slam/data/shader/rainbow.frag
0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, 1.00 ES, and 3.00 ES

error : failed to link program
error: linking with uncompiled/unspecialized shadererror: linking with uncompiled/unspecialized shader
construct solver: lm_var_cholmod
done
error : failed to compile shader /home/mat/catkin_ws/src/interactive_slam/data/shader/rainbow.vert
0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, 1.00 ES, and 3.00 ES

error : failed to compile shader /home/mat/catkin_ws/src/interactive_slam/data/shader/rainbow.frag
0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, 1.00 ES, and 3.00 ES

error : failed to link program
error: linking with uncompiled/unspecialized shadererror: linking with uncompiled/unspecialized shader